SpenceKonde/arduino-tiny-841 #12: Attiny1634 8 MHz has messed up serial output

estaciolima, 2015-08-13:
Hi,

I'm using the recently updated core for the Attiny1634 and I'm seeing some strange serial output when using 8 MHz oscillator. That doesn't happen with 1 MHz oscillator.

Thanks!

SpenceKonde, 2015-08-14:
Have you remembered to do Burn Bootloader after selecting the board and 8mhz oscillator (assuming you're not using one of my pre-bootloaded boards)? Even if you're not using it with Optiboot, you still need to do "burn bootloader" to set the fuses if you want to change the clock speed. It defaults to 1mhz, so if you never did burn bootloader, sketches compiled with 1mhz would work, but ones compiled with 8mhz would run at 1/8th the expected speed (and serial at 1/8th the baud rate, which would result in gibberish when you tried to listen to it). You can test that the speed is right by uploading Blink sketch and using a LED on the appropriate pin, and making sure the delay is right. 

Assuming it's not that, i'd need more info, because this worked last I checked...

What code is producing this error? 
What baud rate?
Internal or external oscillator?

If Internal, are you using a 1634 or 1634R? The internal oscillator on the 1634 (no R) is not well calibrated, and I wouldn't be the least bit surprised if serial was messed up. But I'd expect it to be just as bad at 1mhz. They sell the R version at a small premium with much better cal - that's what I would recommend using with internal oscillator, and it's what I'm using in the boards I sell now

estaciolima, 2015-08-14:
Oh, I'm very sorry, that's propably the case!

SpenceKonde, 2015-08-18:
No further response, so assuming solved. Please reopen if not.

closed 2015-08-18 by SpenceKonde