SpenceKonde/arduino-tiny-841 #24: Modifying avrdude.conf is still necessary

nickiaconis, 2016-03-03:
From README.md:

> Modifying avrdude.conf should no longer be necessary, ever as of 8/22/2015 changes.

This is still necessary, at least on OSX running Arduino 1.6.7 (which is the latest version at time of writing).

How should we revise so that this is known?

SpenceKonde, 2016-03-03:
This is a serious defect which must be corrected, assuming this is an actual bug, not user error.

What were you trying to do that resulted in a failure? 
Can you provide verbose output from the failed upload? 

What programmer have you selected? As noted in the  If you are doing Burn Bootloader, or Upload using Programmer, you must select a programmer with ATTiny Modern after it's name, otherwise it will still use the old version of the avrdude.conf file - this is a limitation of the IDE. 

I cannot reproduce this issue on Windows.

nickiaconis, 2016-03-03:
It is on an upload. Here is the output (minus the wont-fix warnings):

> Sketch uses 3,096 bytes (37%) of program storage space. Maximum is 8,192 bytes.
> Global variables use 213 bytes (41%) of dynamic memory, leaving 299 bytes for local variables. Maximum is 512 bytes.
> avrdude: AVR Part "attiny841" not found.
> 
> Valid parts are:
>   uc3a0512 = AT32UC3A0512
>   c128     = AT90CAN128
>   c32      = AT90CAN32
>   c64      = AT90CAN64
>   pwm2     = AT90PWM2
>   pwm2b    = AT90PWM2B
>   pwm3     = AT90PWM3
>   pwm316   = AT90PWM316
>   pwm3b    = AT90PWM3B
>   1200     = AT90S1200
>   2313     = AT90S2313
>   2333     = AT90S2333
>   2343     = AT90S2343
>   4414     = AT90S4414
>   4433     = AT90S4433
>   4434     = AT90S4434
>   8515     = AT90S8515
>   8535     = AT90S8535
>   usb1286  = AT90USB1286
>   usb1287  = AT90USB1287
>   usb162   = AT90USB162
>   usb646   = AT90USB646
>   usb647   = AT90USB647
>   usb82    = AT90USB82
>   m103     = ATmega103
>   m128     = ATmega128
>   m1280    = ATmega1280
>   m1281    = ATmega1281
>   m1284p   = ATmega1284P
>   m1284rfr2 = ATmega1284RFR2
>   m128rfa1 = ATmega128RFA1
>   m128rfr2 = ATmega128RFR2
>   m16      = ATmega16
>   m161     = ATmega161
>   m162     = ATmega162
>   m163     = ATmega163
>   m164p    = ATmega164P
>   m168     = ATmega168
>   m168p    = ATmega168P
>   m169     = ATmega169
>   m16u2    = ATmega16U2
>   m2560    = ATmega2560
>   m2561    = ATmega2561
>   m2564rfr2 = ATmega2564RFR2
>   m256rfr2 = ATmega256RFR2
>   m32      = ATmega32
>   m324p    = ATmega324P
>   m324pa   = ATmega324PA
>   m325     = ATmega325
>   m3250    = ATmega3250
>   m328     = ATmega328
>   m328p    = ATmega328P
>   m329     = ATmega329
>   m3290    = ATmega3290
>   m3290p   = ATmega3290P
>   m329p    = ATmega329P
>   m32u2    = ATmega32U2
>   m32u4    = ATmega32U4
>   m406     = ATMEGA406
>   m48      = ATmega48
>   m48p     = ATmega48P
>   m64      = ATmega64
>   m640     = ATmega640
>   m644     = ATmega644
>   m644p    = ATmega644P
>   m644rfr2 = ATmega644RFR2
>   m645     = ATmega645
>   m6450    = ATmega6450
>   m649     = ATmega649
>   m6490    = ATmega6490
>   m64rfr2  = ATmega64RFR2
>   m8       = ATmega8
>   m8515    = ATmega8515
>   m8535    = ATmega8535
>   m88      = ATmega88
>   m88p     = ATmega88P
>   m8u2     = ATmega8U2
>   t10      = ATtiny10
>   t11      = ATtiny11
>   t12      = ATtiny12
>   t13      = ATtiny13
>   t15      = ATtiny15
>   t1634    = ATtiny1634
>   t20      = ATtiny20
>   t2313    = ATtiny2313
>   t24      = ATtiny24
>   t25      = ATtiny25
>   t26      = ATtiny26
>   t261     = ATtiny261
>   t4       = ATtiny4
>   t40      = ATtiny40
>   t4313    = ATtiny4313
>   t43u     = ATtiny43u
>   t44      = ATtiny44
>   t45      = ATtiny45
>   t461     = ATtiny461
>   t5       = ATtiny5
>   t84      = ATtiny84
>   t85      = ATtiny85
>   t861     = ATtiny861
>   t88      = ATtiny88
>   t9       = ATtiny9
>   x128a1   = ATxmega128A1
>   x128a1d  = ATxmega128A1revD
>   x128a1u  = ATxmega128A1U
>   x128a3   = ATxmega128A3
>   x128a3u  = ATxmega128A3U
>   x128a4   = ATxmega128A4
>   x128a4u  = ATxmega128A4U
>   x128b1   = ATxmega128B1
>   x128b3   = ATxmega128B3
>   x128c3   = ATxmega128C3
>   x128d3   = ATxmega128D3
>   x128d4   = ATxmega128D4
>   x16a4    = ATxmega16A4
>   x16a4u   = ATxmega16A4U
>   x16c4    = ATxmega16C4
>   x16d4    = ATxmega16D4
>   x16e5    = ATxmega16E5
>   x192a1   = ATxmega192A1
>   x192a3   = ATxmega192A3
>   x192a3u  = ATxmega192A3U
>   x192c3   = ATxmega192C3
>   x192d3   = ATxmega192D3
>   x256a1   = ATxmega256A1
>   x256a3   = ATxmega256A3
>   x256a3b  = ATxmega256A3B
>   x256a3bu = ATxmega256A3BU
>   x256a3u  = ATxmega256A3U
>   x256c3   = ATxmega256C3
>   x256d3   = ATxmega256D3
>   x32a4    = ATxmega32A4
>   x32a4u   = ATxmega32A4U
>   x32c4    = ATxmega32C4
>   x32d4    = ATxmega32D4
>   x32e5    = ATxmega32E5
>   x384c3   = ATxmega384C3
>   x384d3   = ATxmega384D3
>   x64a1    = ATxmega64A1
>   x64a1u   = ATxmega64A1U
>   x64a3    = ATxmega64A3
>   x64a3u   = ATxmega64A3U
>   x64a4    = ATxmega64A4
>   x64a4u   = ATxmega64A4U
>   x64b1    = ATxmega64B1
>   x64b3    = ATxmega64B3
>   x64c3    = ATxmega64C3
>   x64d3    = ATxmega64D3
>   x64d4    = ATxmega64D4
>   x8e5     = ATxmega8E5
>   ucr2     = deprecated, use 'uc3a0512'

I'm using an AVRISP mkII for uploading, and I had `AVRISP mkII` selected as my programmer. Now that I'm looking, there is no `AVRISP mkII (ATtiny Modern)` programmer. Selecting `AVR ISP (ATtiny Modern)`, which is the closest available, results in:

> Sketch uses 3,096 bytes (37%) of program storage space. Maximum is 8,192 bytes.
> Global variables use 213 bytes (41%) of dynamic memory, leaving 299 bytes for local variables. Maximum is 512 bytes.
> avrdude: ser_open(): can't open device "/dev/cu.usbmodem1a1521": No such file or directory
> Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

So something seems wrong with that.

Also, I had overlooked this part of the README tonight, since it wasn't there last time I was interfacing with this project:

> When uploading sketches via ISP, due to limitations of the Arduino IDE, you must select a programmer marked ATTiny Classic or ATTiny Modern from the programmers menu (or any other programmer added by an installed third party core) in order to upload properly to most parts.

Maybe the line that states modifying avrdude.conf is no longer necessary can point people to the ISP instructions?

SpenceKonde, 2016-03-03:
Okay, I'll look over that part of the instructions. 

The AVR ISP mkII (ATtiny Modern) is not enabled, because it's not very common, and people don't like how both of my ATTiny cores add their own blob of programmers to the menu, so I disabled the uncommon programmers. 

You can enable it by uncommenting these lines in programmers.txt (by removing the # at the start of each line )

`
#avrispmkii.name=AVRISP mkII  (ATtiny Modern)
#avrispmkii.communication=usb
#avrispmkii.protocol=stk500v2
#avrispmkii.program.protocol=stk500v2
#avrispmkii.program.tool=avrdude
#avrispmkii.program.extra_params=-Pusb
`

nickiaconis, 2016-03-10:
Would it be possible to offer a third library that contains only the programmers? (And remove the programmers from the two ATtiny libraries?) That way, you can include all if then without taking up too much space. The downside being people need two libraries to get started. Also, I'm not sure if there's a difference between the programmers in your two ATtiny cores.

SpenceKonde, 2016-03-13:
I will consider that - My concern is that it might present difficulties when people install it without reading the documentation about needing a second library for programmers....

SpenceKonde, 2016-03-13:
The other approach would be to say "okay, it's time to actually merge these fucking cores already". 

Could I take a major copout w/regards to merging them by doing the following? 

In one of the cores, rename avr/cores/tiny to avr/cores/tinymodern (for example), and adjust boards.txt appropriately. Then just copy the other core's avr/cores/tiny to that one, and concatenate the boards.txt entries... 

That would work right?

SpenceKonde, 2016-03-13:
Okay, made a new issue for this matter:
https://github.com/SpenceKonde/arduino-tiny-841/issues/30

closed 2016-03-13 by SpenceKonde