SpenceKonde/arduino-tiny-841 #25: Warnings when compiling sketch

nickiaconis, 2016-03-03:
Running Arduino 1.6.7 on OSX, I see the following warnings when compiling for ATtiny841:

> Warning: platform.txt from core 'ATtiny Modern' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

> Warning: platform.txt from core 'ATtiny Modern' contains deprecated recipe.preproc.macros="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} {preproc.macros.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}", automatically converted to recipe.preproc.macros="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} {preproc.macros.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{preprocessed_file_path}". Consider upgrading this core.

I'm assuming the command to which it is automatically converted can directly replace the old command in this codebase, but I haven't investigated further.

SpenceKonde, 2016-03-03:
Fixing those warnings breaks backwards compatibility with IDE version 1.6.5 (which for many users is the last version that works - arduino builder bugs in 1.6.6 and 1.6.7 have resulted in widespread problems that prevent compiling from working). 

The warnings do not interfere with compiling or uploading, and can be safely disregarded. 

Closed, no plans to change.

closed 2016-03-03 by SpenceKonde

nickiaconis, 2016-03-03:
Fair enough. Thanks.