SpenceKonde/arduino-tiny-841 #28: HardwareSerial.h doesn't define serial config values

fson, 2016-03-11:
Serial config values (`SERIAL_8N1`, `SERIAL_5N2` etc) passed as the second (optional) argument to `Serial.begin(speed, config)` (and some other libraries like modbus-arduino) are missing from this core.

In the official core these are defined in: https://github.com/arduino/Arduino/blob/26e25a4d58ff1a970d4790e5ccc3a5f4c3dea398/hardware/arduino/avr/cores/arduino/HardwareSerial.h#L67-L91

SpenceKonde, 2016-03-11:
More serious than that - it doesn't accept the second argument either!

SpenceKonde, 2016-03-11:
Try it now (github version, ofc). I just took a quick stab at pulling in those changes.

fson, 2016-03-11:
Thanks, that was fast! At least my sketch compiles now, can't test more than that until I have the master device setup.

fson, 2016-03-11:
Hold on, it didn't actually compile yet (I had wrong board selected when testing with the new code). Here's what I had to change to make it compile: https://github.com/fson/arduino-tiny-841/commit/3706d58fa8fb87524c8e280820229a5cc3eee8c1

SpenceKonde, 2016-03-11:
Oops, yeah, missed a couple.

closed 2016-03-11 by SpenceKonde

fson, 2016-03-11:
@SpenceKonde `&UCSR0c` should be `&UCSR0C` (capital C) [here](https://github.com/SpenceKonde/arduino-tiny-841/blob/19c7bd8e90ec0ac27267edc2afb0cadf03b5d945/avr/cores/tiny/HardwareSerial.cpp#L286). Otherwise good now!

commit 19c7bd8e (2016-03-11): Fix #28

Missed a few